Field-Robotics-Lab/dave #277: In the folder `uuv_ws/src`, this project cloned  `acoustic_msgs` package from https://github.com/apl-ocean-engineering/marine_msgs/tree/main/marine_acoustic_msgs to the folder `hydrographic_msgs`. But unfortunately, the package has been updated, NOT compatible. If you just want to run codes, the simplest way is the rollback of the downloaded package.

huangbintao188, 2023-11-09:
In the folder `uuv_ws/src`, this project cloned  `acoustic_msgs` package from https://github.com/apl-ocean-engineering/marine_msgs/tree/main/marine_acoustic_msgs to the folder `hydrographic_msgs`. But unfortunately, the package has been updated, NOT compatible. If you just want to run codes, the simplest way is the rollback of the downloaded package.
```
cd uuv_ws/src/hydrographic_msgs
git checkout 5344060071d0a72857a9efb558cc1f2d7c0052c6
```

_Originally posted by @CCLings in https://github.com/Field-Robotics-Lab/dave/issues/272#issuecomment-1678367126_

huangbintao188, 2023-11-09:
After going back, run the example. My sonar has no image, and it is 0 straight when viewed with rqt.

mabelzhang, 2023-11-09:
Hi, can you please edit the title and the description so that there's more context to your question?
It is unclear what you're trying to run. The post that you are quoting is about compiling, not about running anything.

huangbintao188, 2023-11-10:
I used the git checkout 5344060071d0a72857a9efb558cc1f2d7c0052c6 method and catkin_make succeeded, when i try to run roslaunch nps_uw_multibeam_sonar sonar_tank_blueview_p900_nps_multibeam.launch (https://field-robotics-lab.github.io/dave.doc/contents/dave_sensors/Multibeam-Forward-Looking-Sonar/) .My gazebo interface starts and runs normally, but the other window buleview_p900 is dark and has no sonar image. Check the configuration artificialVehicleVibration and it is true. Open rqt to check the sonar image and the values are all 0, which means it is not a computer problem. It may be some configuration error, but I No reason can be found at present. May I ask what is the reason for this and how did you solve it?

woensug-choi, 2023-12-04:
https://github.com/Field-Robotics-Lab/dave/pull/269 and https://github.com/Field-Robotics-Lab/nps_uw_multibeam_sonar/pull/55 would have solved this. If the problem persists. let me know!

closed 2023-12-04 by woensug-choi