Field-Robotics-Lab/dave #278: no sonor image

huangbintao188, 2023-11-10:
I used the git checkout 5344060071d0a72857a9efb558cc1f2d7c0052c6 method and catkin_make succeeded, when i try to run roslaunch nps_uw_multibeam_sonar sonar_tank_blueview_p900_nps_multibeam.launch [https://field-robotics-lab.github.io/dave.doc/contents/dave_sensors/Multibeam-Forward-Looking-Sonar/](url) .My gazebo interface starts and runs normally, but the other window buleview_p900 is dark and has no sonar image. Check the configuration artificialVehicleVibration and it is true. Open rqt to check the sonar image and the values are all 0, which means it is not a computer problem. It may be some configuration error, but I No reason can be found at present. 
[image]

gaoning0109, 2023-11-15:
Are you solved it ?

huangbintao188, 2023-11-15:
no, I don't know exactly why. It's the same with docker method.

gaoning0109, 2023-11-16:
I found the P_Beams in gazebo_multibeam_sonar_ray_based.cpp is empty. Maybe the cuda cal something wrong.

woensug-choi, 2023-11-29:
Let me get back to you after I finish merging pending PRs first.

huangbintao188, 2023-12-03:
> I found the P_Beams in gazebo_multibeam_sonar_ray_based.cpp is empty. Maybe the cuda cal something wrong.

What is your CUDA version

woensug-choi, 2023-12-04:
@huangbintao188 Checked on empty PBeams on CUDA 11.4.. The original code is developed on CUDA 11.1

woensug-choi, 2023-12-04:
@huangbintao188 Oh! Are you using Docker? or On the Host??
I've tried on Ubuntu 22.04, NVIDIA 545, CUDA 12.3 and it works perfectly fine!
I am guessing mismatch between Docker CUDA version and the host CUDA version.

huangbintao188, 2023-12-04:
Yes, that should be my version mismatch. If you run well on Ubuntu 22, I can configure it on our server host. Does Dave and np_sonar support ROS2 on Ubuntu 22? How is the specific process for deploying Dave? Thank you

huangbintao188, 2023-12-04:
@[woensug-choi](https://github.com/woensug-choi)

woensug-choi, 2023-12-05:
Not for ROS2. I've tested with docker environmont meyhod on Ubuntu 22.04. Sonar worked fine :)

Evintjh, 2023-12-06:
Hi, I am using NVIDIA 535, docker ubuntu 20.04 and cuda version 11.1. But I am also not seeing any sonar images. Anyone knows why? @huangbintao188 @woensug-choi @gaoning0109

woensug-choi, 2023-12-06:
for cuda 11.1 the nvidia 535 isnt compatible. Use 450 which is too old...

[image]

[image]

Evintjh, 2023-12-06:
Oh, but I tried NVIDIA 535, docker ubuntu 20.04 and cuda version 12.3 too and it gave me similar results @woensug-choi

woensug-choi, 2023-12-07:
Ive just tried exact same environment yesterday and it worked. Does your host machine and docker has same cuda versions?? you can check by typing nvidia-ami both inside and outside docker

huangbintao188, 2023-12-07:
I plan to deploy it on a lab computer, it's Ubuntu22, cuda is 12.1, is this environment compatible with Dave's docker method? Do you need me to modify the e.g. docker file? @woensug-choi

woensug-choi, 2023-12-07:
try docker method and see if cuda version matches inside and outside docker environment

Evintjh, 2023-12-09:
Hi @woensug-choi , when you used Ubuntu 22.04, which version of ROS did you use? I thought Noetic is not compatible with it?

Evintjh, 2023-12-10:
[image]
[image]

woensug-choi, 2023-12-11:
folliwing installation instruction on multibeam sonar document page will do

woensug-choi, 2024-01-30:
@Evintjh Any updates?

aimas-lund, 2024-02-27:
@Evintjh upgrading from Nvidia Driver 535, CUDA 12.2 to Driver 545, CUDA 12.3 seemed to solve this issue for me.
Notably I'm running this in the Docker container on Ubuntu 22.04.4

zxylov, 2024-03-21:
@woensug-choi I have followed installation instruction on multibeam sonar document page to configure docker ,but the version of cuda is 12.4,how to change the version to 12.3, or the version of 12.4 can also accept the image ?

Aaronl6, 2024-07-08:
@akgoins thank you very much. Solved the problem following your advice. Rebuild is important