Field-Robotics-Lab/dave #290: Underwater Drag

BALLLIKEPOPI, 2024-06-14:
Hi, I'm working some underwater UAV control, and I am wondering if this simulator can simulate the water drag of the drone which may related to the velocity and the contact surface area of the drone, thanks!

woensug-choi, 2024-06-26:
Yes, the hydrodynamic plugin is used that is inherited from https://uuvsimulator.github.io/packages/uuv_simulator/docs/api/gazebo::UnderwaterObjectPlugin/.

Where to mod: 

https://github.com/Field-Robotics-Lab/dave/blob/e54ea16bdaf7d6dbb466cf141846a03f284a44dd/urdf/robots/glider_description/urdf/glider_hybrid_whoi.xacro#L114-L133

You may also want to use generic plugin for hydrodynamics. Read https://gazebosim.org/api/sim/8/theory_hydrodynamics.html

BALLLIKEPOPI, 2024-06-26:
> Yes, the hydrodynamic plugin is used that is inherited from https://uuvsimulator.github.io/packages/uuv_simulator/docs/api/gazebo::UnderwaterObjectPlugin/.
>
> Where to mod: [...]

Thanks for the reply! I will check it out.

closed 2024-06-26 by BALLLIKEPOPI