Field-Robotics-Lab/dave #292: How can I use Oberon 7 or predator manipulator without a joystick?

SafakYaslak, 2024-10-27:
**I have an ongoing problem with the _REXROV robotic arm in the DAVE project simulation_.** Although I have successfully broadcasted command messages to the topic `/rexrov/oberon7/azimuth/controller/command`, the robotic arm is not responding as expected. When I use the echo command from the terminal, I can see the messages I have sent, I can see the topics correctly via rqt_graph, and I do not receive any error messages.

I broadcast _float64_ messages to the topic `/rexrov/oberon7
[image]
/azimuth/controller/command` at a _frequency of 25 Hz_ using the following command (I have also sent them at different frequencies or default values).

I checked the active nodes using the Rosnode list and confirmed that the node _/rexrov/oberon7/joint_position_controller_oberon7_ is indeed present and expected to control the arm.

I reviewed the subscriptions and publications of the joint position controller node, which confirms that it is subscribed to the necessary topics.

I used rosnode info `/rexrov/oberon7/joint_position_controller_oberon7` to make sure the node is properly connected to the command topic. It broadcasts to` /rexrov/oberon7/azimuth/controller/command` and I verified that it has connections to Gazebo.

**Request for Help**

Please help me how to use the robot arm other than the joystick.

SafakYaslak, 2024-10-27:
[image]

woensug-choi, 2024-10-28:
I believe you would have to make your own node to bride between joystick and arm. (I have not developed the part for robot arm).

SafakYaslak, 2024-10-28:
> I believe you would have to make your own node to bride between joystick and arm. (I have not developed the part for robot arm).

Thank you for your advice. Also, is there an e-mail address or page where I can reach the team that developed the robot arm?

closed 2024-10-30 by SafakYaslak